ste2425/PSP-Bluetooth #8: Psp 3000

Bma36, 2023-04-03:
I want to make it for my psp 3000, is it compatible? do you have twitter to follow your work?

ste2425, 2023-04-03:
It currently doesn't support the 3000 yet. 

The pinout of the various internal connectors on the 3000 is different to the 2000. Sadly i don't have a 3000 to figure out the pinout of these connectors and modify the hardware to make it compatible. 

It would be an easy task to make it compatible but i won't be in a position to purchase a 3000 for a while to investigate.

If you, or anyone else interested in this project, has a 3000 and the ability to open it and figure out that pinout i would happily make the changes needed to support it.

Bma36, 2023-04-03:
Yes, I have a psp 3000 and I can help you by dismantling and taking photos of the connectors, your twitter is restricted, you can't follow it, if you want the connectors' records just say so.

https://twitter.com/Am60691019?t=VmD2-d7xX93HxiJeVbyZvA&s=09

this is my twitter if you can follow it so i can follow back i'll be grateful

Bma36, 2023-04-05:
https://github.com/DobaMuffin/PS2PSP

this is another project to use the ps2 controllers made based on the psp 2000 and was adapted for the 3000, maybe it can help you

Bma36, 2023-04-08:
Up

hpx1984, 2024-05-09:
hi Sir ste2425

how about PSP1000, is it can support too?

Doyle4, 2024-07-19:
From looking at Issues I can see this project is no longer worked on or the dev is not interested in peoples requests, did anyone find a alternative that works on all?

ste2425, 2024-07-19:
Pleas try to avoid passive aggression in order to get a developers attention and/or try an pressure them into working on something for you. 

I am actively working in this and I’d you check my YT you would see somewhat regular uodates on this project. 

I am doing this in my own time for fun. As such I will be working on it in a timescale that’s fits around my personal and family life. 

If you or anyone else for that matter wishes to contribute to help move this project forward I’d happily collaborate with others. 

Right now the ESP is working. You can controll all the PSP’s buttons and I’m 1/3rd the way through designing the PCB to fit in the UMD space. 

I will try and more regularly push to this repo and post YT videos on the project though.

Doyle4, 2024-07-22:
Passive aggressive?
May 9th 2024 most recent issue, then its a whole year a question was asked unanswered, May 2023, not passive aggressive, clear statement to what looked liked an abandoned project.   

Did you run out of bubblewrap to protect yourself from words? Damn…

ste2425, 2024-07-22:
@Doyle4 there is no need to end your comment with sarcasm or just generally being unpleasant. 

Let’s try and keep this civil shall we, this isn’t Twitter. If you are wanting me to continue building something for you to use at no cost to your self that is probably the best way to go about it. 

Like I say if you wish to contribute to this project to move it along faster I will gladly accept PR’s. 

I you wish to help push forward 3000 support open your PSP3000 and send me high res scans of the button FPC’s both sides and back. This is what I lack as I do not own a 3000 yet.

ste2425, 2024-11-07:
Just an update. 

3k should be supported now. For consolizers the latest hardware variant should support it out the box. 

Portables will need custom FPC’s which when I’m finished to firmware I will build. They are just variants on the 2k however so should be very quick to do.

ste2425, 2025-02-19:
Closing this. All variants of the psp 2 and 3k are fully supported .

closed 2025-02-19 by ste2425