ste2425/PSP-Bluetooth #9: Updates for you I filmed a video tutorial!!

Shtonksareup, 2024-06-09:
Hey there @ste2425, 

So I may have gone overboard... haha I just got my set up semi video ready and I really wanted to film a tutorial because 1. I saw people commenting on your short saying it was fake since their is no tutorial, which is dumb but what ever. 2. I was looking for this exact project for what felt like ages, and every reddit post claims its impossible.... well clearly not. I want to make sure you get all the credit for your work though. I just put the thing together and filmed it. I will add a google drive link to the video so you can watch it. Ignore the audio for now, I plan to put graphics on it and dub over my voice for directions. Since I do a full psp teardown and the install. I want people to be able to skip around the tutorial as easy as possible. The only thing I will say in the video that is feedback on the current BOM is that the Right 14pin connector cable I think should be opposite sided, not the A type. both builds I did I had to get funky with the cable to get it to fit in right, but the opposite side one would let you do two angled bends and fit the right way :). I will explain that in the video the rest will literally just be a video form of your build guide with a little extra direction and help for people who are visual. 

I have a small YT channel with my brother where we play games, but I would not be upset if you felt like the video should be posted on your channel since you did all the work :D you take a look at the video and let me know what you think. 

PS ( I had to watch a YT video on how to submit an issue, for the first time ever I know how my parents feel when I help them with their email or other various tech issues haha) 

Here is the link for the beta version of the video 
[Video Link](https://drive.google.com/file/d/17oPPmgQ6rXEj_2d35QBCag0iO8QnlPM8/view?usp=sharing)

I am also working on a potential shell the can be printed and will cover the system, that will take more time since I have to wait for my prints to finish and then test and correct. But I have a cool design i am starting to like :D

Shtonksareup, 2024-06-10:
@ste2425 Not a problem at all! I am coming off a super busy period myself and starting Wednesday I have am taking a bunch of time off since I am a teacher. I figure you will respond when you can so no rush on that kind of stuff. 

I have always loved taking things apart so I wanted to make videos about that, but just don't really have the time haha! This just felt like the perfect thing to film especially with there not being one out there yet. 

Don't change what your doing at all! I think splitting them into two projects makes the most sense, but I honestly wouldn't do anything else with the BT project in terms of soft ware. It is fun you say your a software guy and hardware isnt your thing I tell my friends the exact opposite. It was so nice for me that all I had to do is plug the Arduino in and hit run. That is honestly what most people would want. Thats why I don't think you need to change anything or offer additional features to the "external BT build" All I wanted as PS controller button support so I could play a few of the PSP games I love on the big screen. With out getting to technical and off track. I think a lot of people aright now are getting home set ups with upscalers, that's the reason I even wanted to do the build in the first place. 

Back on track, I think if you split it into two things and we add the video link/ stl files for the new external shell ( its super basic but adds a little protection. Then leave that project alone. I have played for around 10 hours or so on and off over the past few weeks and I have only ever had it desync once on me and all i did was press the button on the PCB and i was playing again. So I am sure there are people who would want more from you and there are more features that people would want but put all that work into the new project. The old one can be for people like me who have a psp with a busted screen, of in my case I had a MOB that no matter how many different 2000 umd drives I tried it still wouldn't read disks. So something on the board was messed up and I held onto if for parts, but it was the perfect board for this build. Plus if the video helps some people build it and it catches any steam I am sure someone else would pop in and mess around with the Arduino if they really wanted to. 

I watched your video about the current updates, and I think there is a whole group of people that would like the new build you are working on and the fact that it would work for all models would be great. (There are lots of comments asking for that im guessing cause people want to use what ever psp they already have) This project though wasnt to expensive. I bought enough parts for 2 builds and with the tv cable and the second PSP i bought it cost me like $200 USD I have a break down in the graphics I am adding to the video. I think for a bit over $100 a lot of people will want to do this build. The modding community is huge, and some mods are way to expensive. I did one on my original Nintendo a few weeks ago and the parts alone cost me $120. Im getting off topic again though sorry. 

Summing up. You should work on what makes you excited. Its a fun project so don't change what your cause I have time put into the old project. I think they both have uses, and I think people will enjoy building them both or just one or the other for their needs. If they want plug and play and have a broken PSP like me then they are in luck. If they want way more freedom, and control buttoned down into the old shell then you have that in the works and they could help/ keep tabs on the project! you keep doing what you are doing. I think the split into a finished or released version of the external build and then a in the works internal build would be perfect. 

I will stop going on and on now haha. I am away for the next day and a half, but I am hoping to finish the video by Friday, I had to do some final reprints last night and I want to film the extra shell assembly.

Shtonksareup, 2024-06-10:
@ste2425 I got the chance to get the shell printed, so use this video link, I will check my email to make sure I approve your request this time. [Video with the shell at the end](https://drive.google.com/file/d/1I4zoUzjQbjUgBxGIB0SeaWKbQubcTaRH/view?usp=sharing)

Shtonksareup, 2024-06-13:
Alright @ste2425 Here is the link to the finished video, I sent an email over to you, but Im not sure if they made it so I am adding it here for you to take a look at. Make sure I got your name right! :D 
[Video with all finished audio](https://drive.google.com/file/d/1l1CTqXvrykbftFFA1gyGjSH0_6XYXhdx/view?usp=sharing)

ste2425, 2024-06-13:
hey @Shtonksareup sorry I've been meaning to respond the past couple of days.

The video is fantastic. Super good quality and in-depth.

I think i will publish a new release with your version of the mount stl as it looks allot better than what ive built.

Really interesting point about the firmware upgrade and TVout, i never considered that and certainly something to be aware of.

Are you on twitter/discord/YT? Would you be open for a DM to chat about this project? I'll be easier to get hold on on one of those platforms.

Shtonksareup, 2024-06-13:
@ste2425 Thank you so much for saying all of that I really appreciate it!! Yeah I am on discord and youtube I probably check discord more though if you give me your discord name I will add you mine is zeldandlink that should work. :D DM would be way easier.

ste2425, 2024-06-14:
@Shtonksareup Neat, ive ive done it right i should have sent a request over to you. Or if not some random internet person!

Shtonksareup, 2024-06-14:
@ste2425 I think I added you. I sent a little message :) if you didnt get one then I added some one else 👀

closed 2024-07-16 by Shtonksareup