ste2425/PSP-Bluetooth #12: Pressing X Puts the PSP to Sleep?

Grae87, 2024-11-04:
Hi, sorry if this is something basic, but everything seems to work except the X button. When I press it, the console goes to sleep.

I've checked continuity with all the points. I can't find any bridging or cold solder points.
No issues came up with the Arduino flash either.

Can anyone help?

ste2425, 2024-11-04:
The FFC which controls the PSP’s power button is on the same connector as the X button (and all the others on that side for that matter) 

My first port of call would be to verify the FFC is correctly seated at both ends. Also ensure is is not obviously damaged. 

When building this many times I had nudged one of the cables causing behaviour like this. 

Once one of them had an internal break causing similar behaviour but that seems rarer from my experience

ste2425, 2024-11-26:
Closing this as stale. Please re-open if any new information is available.

closed 2024-11-26 by ste2425