ste2425/PSP-Bluetooth #15: How to program from localhost?

Darthagnon, 2024-12-14:
The current implementation looks like it depends on https://ste2425.github.io/ to program the controller - is it possible to program instead from a website running off my laptop, or via USB? I would prefer not to depend on a website that will be 404 in a few years to program hardware.

ste2425, 2024-12-14:
Thanks for the interest in my project!

First of all https://ste2425.github.io/ is me, it is my GitHub username and all the various URLS off that are my repositories with GitHub pages configured, ‘PSP-Bluetooth’ being this repository. 

This repository contains *everything* needed for the PSP-Bluetooth project to function, the hardware files, firmware source and the web app source. The readme documents this and explains the repository structure. 

So long as this repository exists the web app will exist, so essentially so long as the project exists the web app will!

hopefully that alleviates any concerns you have @Darthagnon. 

However everyone are free, and encouraged, to fork this repository. Make changes and enhancements. 

So if you wish you can fork the repo and follow the instructions in the read me to run the web app locally.

ste2425, 2024-12-16:
As there’s been no response going to consider this problem solved. Please reopen if still having concerns.

closed 2024-12-16 by ste2425

Darthagnon, 2024-12-24:
Thank you for the info! I will try running the website from the repo off localhost. 

I was a bit worried as similar projects that used direct-from-web updates for writing firmware (e.g. [Google Stadia](https://garyodernichts.blogspot.com/2023/01/looking-into-stadia-controller.html), [Henlo for PS Vita](https://github.com/SKGleba/henlo_jb)) did not work offline and needed extra work to be suitable for long-term usage.

ste2425, 2024-12-24:
It’s also worth adding for programming the web app is purely a convenience. 

You can download the binaries from the release and program the ESP32 using your referred programming method.