ste2425/PSP-Bluetooth #17: コントローラーの右スティックの斜め入力

niasuniasu, 2025-01-25:
コントローラーの右スティックをD-Padに設定した時、斜め入力が出来ないのですが、出来るようになりますか？

ste2425, 2025-02-02:
This is something i was hoping i would get away with not being an issue.

Lets say the range of the analog stick is -250 - 250. To trigger a button press it has to be over 20 from the end of its range (so -230 or 230).

Trouble is because the analog stick gate is a circle if the stick is in the top left diagonal the left value may be something like -200 and the top value 200. So they are outside the threshold to trigger a value.

I could lower the threshold but that would make the analog stick feel too sensitive for some.

I need to have a think, i have an idea on how to detect if its a diagonal.

Jonnicom, 2025-02-03:
I am glad to see both this and the other issue that you combined with an earlier submitted one are at least on the radar.  I agree, the right stick works ok as it is, but it would be nice to see some type of diagonal integration eventually.  Thanks for all you do, this is a great project!

ste2425, 2025-02-04:
The pre-release build 2.2.0-rc2 has a fix for this implemented.

I have lowered the thresholds slightly and diagonals work in my testing.

It feels ok to me but that is very subjective.

Feel free to give it a try (can be installed from the web app using BLE). 

I will consider this resolved if i don't hear otherwise :)

thanks.

Jonnicom, 2025-02-05:
> It feels ok to me but that is very subjective.

Subjective indeed!  It's going to be different for everyone, and perhaps even game-to-game, just depending on how the player is trying to implement it.

I have my right stick on a Dualsense mapped to the PSP Dpad, and after testing it in Metal Gear Solid: Portable Ops, it works, but it is very picky.  Have to be pointing precisely at the 45 degree angles, or you'll still just get up or right/up or left, etc.

But, in practice it is working and seems possible.  I guess adjustments can be tested from here on out.  I wonder if there is any way to implent the user being able to adjust this sensitivity on their own in the future.  But, I am also aware that we're essentially mapping 4 buttons in an attempt to "fake" 4 additional directions.

I'll keep testing it any other games I can think of in the next few days and give more feedback to try to help out!

ste2425, 2025-02-05:
Thanks for testing this out so quick!

Yea I was on the fence if this would be acceptable or not. Decided to wait for use feedback.

It would be good to have it user controllable.

Need to try and set sensitivity for the four directions and the four diagonals seperatIy i think.

Lowering the threshold any more to account for diagonals I think will reign the feel of  the four main directions if that makes sense.

I think rather than over engineering at this stage I will make it configurable. If it turns out that isn’t enough to make both the  diagonals and the four main directions comfortable then we will have to rethink.

ste2425, 2025-02-07:
Closing as fixed in version 2.2.0.

There will be enhancements to this in https://github.com/ste2425/PSP-Bluetooth/issues/19

Specifically to make the diagonals work better.

closed 2025-02-07 by ste2425