ste2425/PSP-Bluetooth #19: Add extra config options

ste2425, 2025-02-07:
Add a UI to configure the following things

* Power LED brightness
* Threshold to trigger button press when controllers with analog triggers are  mapped to buttons.
* Threshold to trigger button press when analog stick is mapped to buttons.
  * This includes dealing with diagonals better.

Jonnicom, 2025-02-08:
Very much looking forward to the work on the right stick and thresholds, etc!  The current build did enable it, but I don't think it's sensitive enough.  It is extremely specific in Metal Gear Solid Portable Pps, and doesn't work at all how you'd want it to when mapped to the face buttons for a first person shooter like Ghost in the Shell.  But, that's just what I've found in my testing so far.  Glad it's on the radar!

ste2425, 2025-02-14:
@Jonnicom FYI there is a prelease version 2.3.0-rc4. 

Along with a new settings page in the web app. 

If you have the time maybe you could give it a try. It allows a number of different settings to be toggled, analog stick threshold being one of them. 

It is very experimental so you may encounter issues though!

Jonnicom, 2025-02-15:
@ste2425 Tested/Testing now, but yes, wow, what a difference lowering the threshold makes!  Feels closer to real analog input than ever before.

It's still obvious that it's a "gated" 8 direction input, but, I think that's the best you can get considering it was never true analog directional control to begin with.  Still much prefered to use this than using the 4 face buttons to move a camera, especially in first person shooters.  In fact, when in quick movement and motion, you almost can't *tell* that it's gated 8 directions, it does feel rather fluid with the threshold all the way down.  Great job!  I also tested the trigger sensitivity using a dualsense, I can't really feel much of a difference either way myself, but I'm not sure if the dualsense even uses analog triggers or if I'd notice either way!

Out of curiosity, is the analog threshold setting for adjusting just the right stick, or both left and right at the same time?

I don't see any change in the LED brightness, however - not a huge issue, may vary from controller to controller as well. 

Sidenote, made a video reviewing the PS Placeable project, singing high praises - thought you may wanna see it, and I didn't know where else to send it to you!  https://youtu.be/HCmnLQwp4ZM?si=6qqm9faSwKQYEQ1I

ste2425, 2025-02-15:
@Jonnicom  Just watched your video and it was very impressive. Its great to see this project out there in the wild being used and enjoyed by people :)

I do have a couple of questions. 

* i noticed that on the controller mapping screen the images of the PSP and the Controller look odd, washed out. Do you have an special settings in your browser/operating system for high contrast mode or something similar? This is how id expect it to look.

[image]

However in your video it looked like this:

[image]

Would be good to get that fixed!

* Other question you mention it got into a confused state where it thought the PSP on when it was off etc. If you have a chance could you log an issue and add any info how you got into that state? I would like to investigate just to be sure.

Back to this issue specifically thats great news its working for you, could you share the value you set the analog stick threshold to and ill set that as the default.

I have pushed a new RC version which allows changing the button to shut down the console. (at this stage its a fixed alternative combo. If there's enough demand i can ehance further but that would be somewhat more involved).

Anyway thanks for the feedback and testing, has made fixing these issues much easier!

Jonnicom, 2025-02-18:
Hey!  Sorry it took me a while to get to a PC so I could type up a decent response - bad at typing on a phone.

First, yes, the images you have on your end are correct, and the reason they look that way in my browswer is because I use a high-contrast dark mode plug-in, so that's completely on my end!

As for the on/off switch up issue, I think I ended up figuring out the real issue is with the PPrefs plugin that I use, it uses the home button, which brings up the menu, and freezes the console when it's trying to turn off - that's why I thought the option to use other buttons or button combos to turn the console on and off may be a good option, since so many CFW plugins use weird buttons or hold buttons to turn on and off, etc.  Such as PPrefs using the home button, CFW using the select button, and CWcheat using the select button held down for a few seconds to activate.

ste2425, 2025-02-18:
@Jonnicom not a problem thanks for getting back to me.

So im just testing a few more changes. One is an additional shutdown combo instead of the home button like you mention.

The other is a low battery indicator for controllers.

Once done i think this release will be ready.

ste2425, 2025-02-18:
Marking this as complete. Version 2.3.0 should now resolve all the issues mentioned.

closed 2025-02-18 by ste2425