ste2425/PSP-Bluetooth #21: Trouble flashing firmware

schweaty, 2025-03-01:
I'm having a lot of issues flashing the firmware to the board.  I can't get the computer to recognize it.   I've tried at least 15 usb cables at this point.  All I have access to are Mac's.  Either there is a compatibility problem with Mac's or the board is DOA.  This is extremely frustrating.

ste2425, 2025-03-01:
Which browsers are you trying to use? 

Unfortunately I don’t own a MAC however I would expect any chromium based browser to detect the board regardless of the underlying OS. 

One way to rule out the web app/browser being the issue would be to try and program it via the Arduibo IDE. Which I understand is a pain having to setup the ESP32 core but that way you can be certain that if that doesn’t work you have a hardware issue. 

If it is a hardware issue it’s possible it could just be the USB to Serial adapter. 

On my version of the boards I expose the RX and TX pins of the ESP32 to allow for external serial adapters. I don’t know if Dan has kept those pads accessible (assuming it’s one of Dans boards you have) but that is another option.

schweaty, 2025-03-01:
I'm using chrome, but I dont think it's a browser issue.  The computer (a MacBook Pro and old iMac) just are not seeing the board when I plug it in via usb.

ste2425, 2025-03-01:
Ok. Are you sure all your USB cables a data and not just power? I know you said you tried multiple but just to be sure so you have any other device you can test the cables with?

If you have either a USB -> Seriwl adapter or an old Arduino Uno you can bypass the onboard USB to Serial chip just incase that is at fault.

schweaty, 2025-03-01:
I just checked mist of the cables using an Xbox controller.  At least half of them do carry data.  None of them connected the blutooth board to the computer.  I do not have a USB to serial or arduino uno.

schweaty, 2025-03-01:
It just occurred to me I could use my android phone with a USB hub in desktop mode.  Plugged in the USB and it did not detect it.  I got another error message in chrome about serial web not ebabled so I don't think I could have lashed it even if it was detected.

ste2425, 2025-03-02:
Interesting, if you haven’t enabled it yet coukd you share pictures of what the page looks like?

It should do feature detection and detect if you have Web Serial enabled. It should disable the buttons in the UI if not to prevent you even trying.

schweaty, 2025-03-02:
The Serial option under program was disabled in chrome.  My initial conclusion is that the only way to flash th8s board is on a windows PC

ste2425, 2025-03-02:
Ok. Closing this as it’s a known limitation of web serial that it is a very experimental technology with only Chromium based support. 

However Chrome not working on MacOS is a little surprising (or maybe. It considering Safari’s stance is against the technology). 

As I don’t have a Mac I can’t investigate much. I’ll look into trying to detect if the User is using a Mac and show a message explaining the situation. 

Besides that though I think there is not allot I can do. 

I may add a Manual programming option. If someone has no choice but to use Mac you can download the binary and flash it via the Arduino IDE. That would be an option for Mac only users as a work around.

closed 2025-03-02 by ste2425