ste2425/PSP-Bluetooth #22: Unable to flash

salt86, 2025-06-14:
I can't seem to flash the board through USB. I'm getting this error

```
Downloading binary...
Downloaded, size: 879136
Getting connection to ESP...
esptool.js
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting...
AggregateError: All promises were rejected
```

Also using the serial monitor outputs this

`NetworkError: Failed to execute 'open' on 'SerialPort': Failed to open serial port.`

I am using Chrome on Linux. I did attempt using Chromium, but couldn't find any serial ports as easily.

ste2425, 2025-06-16:
@salt86 sorry been trying to respond to this all weekend and it refused to load comments.

Anyway glad you found a work around. I think ill add in the wizard a note saying you can manually download the binaries and flash using any ESP32 flasher tool people prefer. Just in case the webapp refuses to work (the tech is experimental sadly).

Anyway what distro were you running? I tested it on Ubuntu right at the start of the project and it worked for me then, however its possible something changed since then.

If you have the time could you open the console in your browsers dev tools and try again? Its possible it may log more info there that my error handling isn't catching. (Note dev tools needs to be open before trying to flash it otherwise logs may be missed).

I'll try and repro it my end and see if there is anything that can be done.

salt86, 2025-06-17:
Unfortunately, I had some trouble getting the ribbon cables clamped in to the module due to my breaking the ribbon cable clamps. That being said, I'm afraid to take the PSP out of its shell. I will post my specs though:

```
$ uname -a
Linux Blue 6.11.0-26-generic #26~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Apr 17 19:20:47 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```
i also know that it was detected via `lsusb`

ste2425, 2026-10-27:
I have successfully flashed via USB using Fedora and Chromium. 

It's possible in the time since browser updates may have resolved it. USBSerial is an experimental web technology which i believe, even now, not all browsers have agreed to support.

Going to close this as nobody else has reported this. 

If you still face issues, or if others are facing this issue, please re-open.

closed 2026-10-27 by ste2425